stm32duino/STM32RTC #9: Weekdays not defined for stm32f1

LieBtrau, 2018-08-11:
Hello,

I tried the epoch example.  It works fine when you run it with the given date.  Replacing the epoch with a current one, i.e. 1534017459 reveals a bug.
The library refuses to set the date.  Setting time works ok.

It starts to go wrong on line 444 of rtc.c:
` if(IS_RTC_YEAR(year) && IS_RTC_MONTH(month) && IS_RTC_DATE(day) && IS_RTC_WEEKDAY(wday)) {`

The `IS_RTC_WEEKDAY(wday)` doesn't return a correct value.
Diving deeper: rtc.h, line 105: `#if defined(STM32F1xx) && !defined(IS_RTC_WEEKDAY)`.  The weekdays for stm32f1xx are defined in this rtc.h, while for the other architectures this is done in the stm32(blablabla)_hal_rtc.h files.  Strange...

A quick and dirty fix is adding the #defines of the weekdays in rtc.h.
`

fpistm, 2018-08-13:
Hi @LieBtrau, nice catch.
I've proposed a Fix #10

LieBtrau, 2018-08-13:
Hi @fpistm, I just uploaded your code to a BluePill.  It works now with the UTC timestamp 1534017459 and with the current UTC timestamp.

PR #10 (2018-08-13): Fix week day issue
RTC_WEEKDAY_SUNDAY definition in STM32F1 HAL is 0 while 7 for other series.

```c
struct tm:
    tm_wday
    The number of days since Sunday, in the range 0 to 6.
```
Other days are aligned to the` tm_wday` value.

Fix #9
commit 1b10c6b7 (2018-08-13): Fix week day issue

RTC_WEEKDAY_SUNDAY definition in STM32F1 HAL is 0 while
7 for other series.

struct tm:
    tm_wday
    The number of days since Sunday, in the range 0 to 6.

Other days are aligned to the tm_wday value.

Fix #9
closed 2018-08-13 by LieBtrau