stm32duino/STM32RTC #11: RTC looses time upon system reset

LieBtrau, 2018-08-13:
Hello,
The RTC is powered from the VBAT pin.  It should be able to keep tracking time, even though the main power supply is off.  According to [this site](http://embedded-lab.com/blog/stm32s-internal-rtc/) that should be possible.

I checked with an oscilloscope if the LSE keeps running if the main VDD is off while VBAT is powered.  That is ok.

The parameter resetTime in the `rtc.begin(bool resetTime, RTC_Hour_Format format = RTC_HOUR_24);` function probably allows to setup the RTC in the application without resetting its current time.  It doesn't have the expected behaviour.

fpistm, 2018-08-13:
This feature is currently not implemented.
https://github.com/stm32duino/Arduino_Core_STM32/issues/266

And yes reset time will be used for that.

LieBtrau, 2018-08-13:
Hi @fpistm, do you have any idea when this feature will be available?

PR #15 (2019-02-26): Handle reset time
Require STM32 core version higher than 1.5.0.

Fix #11 

Require https://github.com/stm32duino/Arduino_Core_STM32/pull/442 to be functional.

This PR do not break backward compatibilities with old core version as new core version API is used to hadnle this.
commit f28864df (2019-02-26): Handle reset

Require STM32 core version higher than 1.5.0.

Fix #11
closed 2019-02-26 by fpistm