stm32duino/STM32RTC #16: With i2c new library doesn't work

charlie1024, 2019-02-28:
Hello,
I recently seen the board library was updated some days ago and the HAL feature was newly implemented.
I thought with the new master branch board platform the new library will work, so with my Nucleo-F411RE I tried to operate the new library.

It worked without I2C. I use D14/D15(I2C1) and if I connect any I2C device, then the MCU goes to forever sleep(all codes are stopped). 

This code shows:
`#include <Wire.h>
#include <STM32RTC.h>
STM32RTC& rtc = STM32RTC::getInstance();

void setup() {
  // put your setup code here, to run once:
  rtc.setClockSource(STM32RTC::LSE_CLOCK);
  rtc.begin(false); // initialize RTC 24H format

  if (!rtc.isTimeSet()) {
    rtc.setTime(0, 0, 0);
    rtc.setDate(1, 1, 2019); // fallback
  }
  pinMode(LED_BUILTIN, OUTPUT);

  Wire.begin();

  Wire.beginTransmission(0x50);
  Wire.write(0);
  Wire.endTransmission();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}`

0x50 means the EEPROM device and I just wrote a waste data to the line, so there must be no problem.
However If I use this code with any I2C device connected, F411RE just halts and not respond for anything(interrupts are all stopped).

Couldn't be the new library used yet? Or is this a bug?

Thank you.

charlie1024, 2019-02-28:
I additionally confirmed that the code works with 1.5.0 release board platform(using previous library-compatible code).
Initially I thought the RTC library would have an issue but without ALL RTC codes the source didn't work. I doubt the core(at master branch) has another hidden problem. Just using I2C caused halt. It's so confusing.

charlie1024, 2019-02-28:
Additionally confirmed that the source caused halt only with Wire library, without STM32RTC and all related codes. It's so confusing. Initially I thought the RTC library has an issue with the master branch board platform but it seems that the core has another problem.

fpistm, 2019-02-28:
Hi @charlie1024 
Thanks for report.
Could you give me more detail about your setup.
Which core version (master repo? official 1.5.0?..) STM32RTC version (I guess this is the master)? Board exact reference (MBxxxxx)? Arduino IDE version?

charlie1024, 2019-02-28:
Hello,
I have MB1136-C-04(It has a 32.768kHz crystal).
Used core was directly got from the master branch. Official 1.5.0 seems doesn't support the new library so It will be out of the problem. STM32RTC version is from the master branch as well.
Finally, I'm using Arduino 1.8.8, which is the latest and before the changing the core the Arduino didn't made any problems.

fpistm, 2019-02-28:
Ok thanks. I will test on my side when possible.