stm32duino/STM32RTC #18: RTC_LSE_CLOCK is not a member of 'STM32RTC'

tuxnull, 2019-07-20:
When trying to use the external quartz on my Blue Pill STM32F103C8T6 by running:
`rtc.setClockSource(STM32RTC::RTC_LSE_CLOCK); `
the compiler returns 
`LEDMatrixClock:46:32: error: 'RTC_LSE_CLOCK' is not a member of 'STM32RTC'
   rtc.setClockSource(STM32RTC::RTC_LSE_CLOCK);
                                ^~~~~~~~~~~~~`

tuxnull, 2019-07-20:
Fixed by renaming RTC_LSE_CLOCK to LSE_CLOCK.
Issue still relevant though, still this has to be changed in the examples.

fpistm, 2019-07-20:
Hi @tuxnull,
Right enum has been changed but in which example you get old name?

tuxnull, 2019-07-20:
I found it in the RTCClockSelection example:

https://github.com/stm32duino/STM32RTC/blob/96464ec25de2a6e4a08180f9ab7c40d590bf724c/examples/RTCClockSelection/RTCClockSelection.ino#L63

closed 2019-07-20 by tuxnull