stm32duino/STM32RTC #22: STM32 RTC with attache battery freeze when power is off

HamidSaffari, 2020-04-18:
Hi, there. I read somewhere that you don't implemented the LSE yet to clocking the RTC from low speed oscillator. and it did not work neither when I even set it to  "rtc.setClockSource(STM32RTC::LSE_CLOCK);" .
But by using this library it works like a charm: https://github.com/ZulNs/STM32F1_RTC 
It allows to use the RTC functionalities of STM32F1xx based boards using external low speed oscillator (LSE). and keep continuing the counting after VDD is out with attached RTC battery.
I just want to inform you that maybe it save you times and use it's code.

fpistm, 2020-04-18:
Hi,
LSE is managed, I don't know where read that.
Anyway, without more information we could not help.
Which core version you used, which board, sketch you used,...?
Any relevant info are wwlcome

HamidSaffari, 2020-04-18:
Oh yes, It's working, sorry man.
I just forget to put "rtc.setClockSource(STM32RTC::LSE_CLOCK);" in my setup;
and by default it was in LSI mode and that was the reason by powering-off the clock stopped although there is still oscillation on 32kHz oscillator.

closed 2020-04-19 by HamidSaffari