stm32duino/STM32RTC #27: Cannot use (emulated) eprom with RTC - hang

angelnu, 2020-06-13:
I am trying to use the stm32duino `eeprom_buffered_write_byte` and `eeprom_buffer_flush` together with this STM32RTC library but it hangs.

Here is the sketch that fails on an NUCLEO LM412KB:

```
#include "arduino.h"
#include "STM32RTC.h"

void setup() {
    delay(3000);
    Serial.begin(115200);
  
    eeprom_buffered_write_byte(0, 0xAA);
    Serial.println(F("BEFORE FLUSH"));
    Serial.flush();
    
    eeprom_buffer_flush();
    Serial.println(F("I NEVER REACH THIS"));
    Serial.flush();

    STM32RTC& rtc = STM32RTC::getInstance();
    
    //If this line is remove then the sketch works as expected
    rtc.begin();
}

void loop(){}
```

If I remove the RTC_INIT call inside STM32RTC::begin then it also work so it mus be there. An it is enough having the code linked for it to affect the eeprom.

fpistm, 2020-06-13:
Which core version you used?

angelnu, 2020-06-14:
I use the latest stm32 platform (4.1) in platformio which pulls stm32duino version 1.8. They have not updated it to use 1.9 and the opencd version is too old for the Nucleo L412KB so I upload the sketch using the STM32CubeProgrammer.

I uploaded the test to https://github.com/angelnu/test_eeprom_l4

Output from platformio with the build:

```
test_eeprom_l4]$ platformio run
Processing nucleo_l412kb (platform: ststm32; board: nucleo_l412kb; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l412kb.html
PLATFORM: ST STM32 6.1.0 > ST Nucleo L412KB
HARDWARE: STM32L412KBU6 80MHz, 40KB RAM, 128KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduinoststm32 4.10800.200207 (1.8.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing CapacitiveSensor
git version 2.17.0
Cloning into '/Users/anunez/Documents/PlatformIO/Projects/test_eeprom_l4/.pio/libdeps/nucleo_l412kb/_tmp_installing-3rv1syb5-package'...
CapacitiveSensor @ 802c3c4 has been successfully installed!
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CapacitiveSensor> 0.5.1 #802c3c4
|-- <CayenneLPP> 1.1.0
|   |-- <ArduinoJson> 6.15.2
|-- <STM32duino RTC> 1.1.0
```

fpistm, 2020-06-14:
Ok. So this is fixed with the 1.9.0.

fpistm, 2020-06-14:
https://github.com/stm32duino/Arduino_Core_STM32/issues/930#issuecomment-584212552

angelnu, 2020-06-14:
Ups, sorry for the duplicated issue ... Confirmed that the fix works when I manually add the alignment.

I will fork the platformio package to use 1.9 until they upgrade it upstream.

closed 2020-06-14 by angelnu