stm32duino/STM32RTC #29: RTC date lost after reset

MasterSch, 2020-06-26:
Hi,
I use vscode ide with platformio and a bluepill.
I tried the Simple RTC example of the library.
After reset or disconnecting the power source the date is set back to 1/1/00 but the time keeps going on.
i changed something in the sketch. After a reset and the init of rtc the first print of the date is wrong.
After that the program sets a new date and time.
i attached the scetch and the platform.ini files. 

[main.txt](https://github.com/stm32duino/STM32RTC/files/4837228/main.txt)

[platformio - ini.txt](https://github.com/stm32duino/STM32RTC/files/4837245/platformio.-.ini.txt)

fpistm, 2020-06-26:
Hi,

First we do not support PIO. Only Arduino IDE. You should use the forum for this kind of issue:
https://www.stm32duino.com

You have to ensure VBAT is always on, and use LSE (it seems it is the case).
And also use the last version of the RTC. I don't know how PIO handle this...

delgadosouza, 2020-06-27:
Same issue here, using Arduino IDE and BluePill STM32F103C8T6.

The time is kept fine after pushing reset. The reset button freezes the time while pressed but when released the time goen on. But the date is lost and goes back to 01/01/00.

It looks like a ST employee said the date is kept in sram:
https://community.st.com/s/question/0D50X00009XkekS/stm32f103-rtc-vbat-date-resets-after-power-off

I am using the LowPower library, but only going to sleep. For uploading new code usind the STLink I have to press the reset button so I noticed the date is reset.

The only time I set the time/date is when I send a comand via serial, so no time/date change in setup.

fpistm, 2020-06-27:
Ah on F1 RTC has a bug, IIRW. You should look the errata sheet.

MasterSch, 2020-06-29:
Hi,
thank you so far. i found my solution in a different library https://github.com/ZulNs/STM32F1_RTC
this works good. date and time keep going on after reset or power off with a battery on VBAT.

closed 2020-06-29 by MasterSch