stm32duino/STM32RTC #30: Library does not work with latest platformio

angelnu, 2020-07-05:
Platformio updated to the latest [platform-ststm32](https://github.com/platformio/platform-ststm32/releases/tag/v7.0.0) from 10 days ago does not work after the library split here (see error output bellow).

The problem is that the stm32duino version is v1.9.0 and this results in rtc.h and rtc.c not being used. I changed the version checks to allow for this version until a new platform-ststm32 is available. The library compiles fine and I do not see any side effects.

I will do a PR since anyway I have to keep a fork until all fixes become available. I have never tried to update platform-ststm32  but this would be another alternative to pick up latest stm32duino versions.

Compile error:
```
In file included from lib/STM32LowPower/src/STM32LowPower.h:49,
                 from src/sleep.cpp:8:
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:77:17: error: 'HOUR_FORMAT_12' was not declared in this scope; did you mean 'RTC_HOURFORMAT_12'?
   77 |       HOUR_12 = HOUR_FORMAT_12,
      |                 ^~~~~~~~~~~~~~
      |                 RTC_HOURFORMAT_12
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:78:17: error: 'HOUR_FORMAT_24' was not declared in this scope; did you mean 'RTC_HOURFORMAT_24'?
   78 |       HOUR_24 = HOUR_FORMAT_24
      |                 ^~~~~~~~~~~~~~
      |                 RTC_HOURFORMAT_24
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:82:12: error: 'HOUR_AM' was not declared in this scope; did you mean 'HOUR_24'?
   82 |       AM = HOUR_AM,
      |            ^~~~~~~
      |            HOUR_24
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:83:12: error: 'HOUR_PM' was not declared in this scope; did you mean 'HOUR_24'?
   83 |       PM = HOUR_PM
      |            ^~~~~~~
      |            HOUR_24
In file included from .pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:39:
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:77:17: error: 'HOUR_FORMAT_12' was not declared in this scope; did you mean 'RTC_HOURFORMAT_12'?
   77 |       HOUR_12 = HOUR_FORMAT_12,
      |                 ^~~~~~~~~~~~~~
      |                 RTC_HOURFORMAT_12
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:87:28: error: 'OFF_MSK' was not declared in this scope
   87 |       MATCH_OFF          = OFF_MSK,                          // Never
      |                            ^~~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:88:28: error: 'SS_MSK' was not declared in this scope
   88 |       MATCH_SS           = SS_MSK,                           // Every Minute
      |                            ^~~~~~
Compiling .pio/build/generic_basic_v1_debug/lib495/MCCI LoRaWAN LMIC library_ID5774/hal/getpinmap_catena4612.cpp.o
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:89:28: error: 'SS_MSK' was not declared in this scope
   89 |       MATCH_MMSS         = SS_MSK | MM_MSK,                  // Every Hour
      |                            ^~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:78:17: error: 'HOUR_FORMAT_24' was not declared in this scope; did you mean 'RTC_HOURFORMAT_24'?
   78 |       HOUR_24 = HOUR_FORMAT_24
      |                 ^~~~~~~~~~~~~~
      |                 RTC_HOURFORMAT_24
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:89:37: error: 'MM_MSK' was not declared in this scope
   89 |       MATCH_MMSS         = SS_MSK | MM_MSK,                  // Every Hour
      |                                     ^~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:82:12: error: 'HOUR_AM' was not declared in this scope; did you mean 'HOUR_24'?
   82 |       AM = HOUR_AM,
      |            ^~~~~~~
      |            HOUR_24
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:83:12: error: 'HOUR_PM' was not declared in this scope; did you mean 'HOUR_24'?
   83 |       PM = HOUR_PM
      |            ^~~~~~~
      |            HOUR_24
Compiling .pio/build/generic_basic_v1_debug/lib495/MCCI LoRaWAN LMIC library_ID5774/hal/getpinmap_catena4617.cpp.o
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:90:28: error: 'SS_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                            ^~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:90:37: error: 'MM_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                                     ^~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:87:28: error: 'OFF_MSK' was not declared in this scope
   87 |       MATCH_OFF          = OFF_MSK,                          // Never
      |                            ^~~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:90:46: error: 'HH_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                                              ^~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.h:88:28: error: 'SS_MSK' was not declared in this scope
[... 489 lines cut ...]
      |                          ^~~~~~~
      |                          HOUR_12
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp: In member function 'void STM32RTC::syncDate()':
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:977:5: error: 'RTC_GetDate' was not declared in this scope; did you mean 'HAL_RTC_GetDate'?
  977 |     RTC_GetDate(&_year, &_month, &_day, &_wday);
      |     ^~~~~~~~~~~
      |     HAL_RTC_GetDate
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp: In member function 'void STM32RTC::syncAlarmTime()':
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:988:5: error: 'hourAM_PM_t' was not declared in this scope
  988 |     hourAM_PM_t p = HOUR_AM;
      |     ^~~~~~~~~~~
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:991:38: error: 'p' was not declared in this scope
  991 |                  &_alarmSubSeconds, &p, &match);
      |                                      ^
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:990:5: error: 'RTC_GetAlarm' was not declared in this scope; did you mean 'HAL_RTC_GetAlarm'?
  990 |     RTC_GetAlarm(&_alarmDay, &_alarmHours, &_alarmMinutes, &_alarmSeconds,
      |     ^~~~~~~~~~~~
      |     HAL_RTC_GetAlarm
.pio/libdeps/generic_basic_v1_debug/STM32duino RTC/src/STM32RTC.cpp:992:26: error: 'HOUR_AM' was not declared in this scope; did you mean 'HOUR_12'?
  992 |     _alarmPeriod = (p == HOUR_AM) ? AM : PM;
      |                          ^~~~~~~
      |                          HOUR_12
*** [.pio/build/generic_basic_v1_debug/lib6f5/STM32duino RTC/STM32RTC.cpp.o] Error 1
==
```

angelnu, 2020-07-05:
Provided PR -> https://github.com/stm32duino/STM32RTC/pull/31

fpistm, 2020-07-07:
As mention is the PR #31 I don't know how PIO manage this anyway there is no issue with the version.
The rtc drivers has been removed from the core after the 1.9.0 not the 1.8.0. So if version is > 1.9.0 then the library have to include the rtc.h.

closed 2020-07-07 by fpistm