stm32duino/STM32RTC #32: times reset after power cycle

paulporto, 2020-07-15:
the time its lost after power cycling, although LSE_CLOCK is selected, and the vbat is connected
im using the epoch example with some modifications to be able to the set time via UART

arduino 1.8.13
stm32f103 bluepill
library.properties:
    name=STM32duino RTC
    version=1.1.0
    author=STMicroelectronics, Wi6Labs
    maintainer=stm32duino ........
```
/*
  Epoch

  This sketch shows how to manage the RTC using Epoch time

  Creation 12 Dec 2017
  by Wi6Labs
  Modified 03 Jul 2020
  by Frederic Pillon for STMicroelectronics

  This example code is in the public domain.

  https://github.com/stm32duino/STM32RTC
*/

#include <STM32RTC.h>
#include <time.h>

/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

long NewEpoch = 0;

void setup() {
  delay(1000);
  Serial.begin(115200);

  // Select RTC clock source: LSI_CLOCK, LSE_CLOCK or HSE_CLOCK.
  // By default the LSI is selected as source.
  rtc.setClockSource(STM32RTC::LSE_CLOCK);

  rtc.begin(); // initialize RTC 24H format

  //rtc.setEpoch(1094747232); // 
}

void loop() {
  uint32_t ss = rtc.getSubSeconds();
  uint32_t epoch = rtc.getEpoch();
  time_t rawtime = epoch;
  struct tm ts;
  char buf[80];

  Serial.print("Unix time = ");
  Serial.println(epoch);

  Serial.print("Seconds since Jan 1 2000 = ");
  Serial.println(rtc.getY2kEpoch());

  // Format time, "ddd yyyy-mm-dd hh:mm:ss zzz"
  ts = *localtime(&rawtime);
  strftime(buf, sizeof(buf), "%a %Y-%m-%d %H:%M:%S", &ts);
  Serial.print(buf);
  Serial.print(".");
  print2digits(ss);
  Serial.println();

  //the following code is for setting time via UART
  long t = millis();
  while(millis()-t < 3000) {
    delay(100);
    while(Serial.available()){
            
      char c = Serial.read();
      if(c == 'e' and NewEpoch>1094747232){
        rtc.setEpoch(NewEpoch);
        Serial.print("> ");
        Serial.println(NewEpoch);
        NewEpoch = 0;
        while(Serial.available()){ Serial.read(); }
        return;
        }
      NewEpoch *= 10;
      NewEpoch += String(c).toInt();
      }
    }

  //delay(10000);
}

void print2digits(uint32_t number) {
  if (number < 100) {
    Serial.print("0");
  }
  if (number < 10) {
    Serial.print("0");
  }
  Serial.print(number);
} ```

paulporto, 2020-07-17:
this person make it work with mbed. i dont understand the code but i will paste it here in case some one find it useful
http://embedded-lab.com/blog/stm32s-internal-rtc/
https://libstock.mikroe.com/projects/view/1742/stm32-s-internal-rtc

i think it may be related to this:
#define enable_backup_module(mode)                        RCC_APB1ENRbits.BKPEN = mode
#define enable_power_control_module(mode)             RCC_APB1ENRbits.PWREN = mode

josh-gaby, 2020-09-24:
The epoch includes date data which is lost on power cycle on stm32f1xx boards, this should be fixed by my pull request [here](https://github.com/stm32duino/STM32RTC/pull/41)

paulporto, 2020-10-31:
> The epoch includes date data which is lost on power cycle on stm32f1xx boards, this should be fixed by my pull request [here](https://github.com/stm32duino/STM32RTC/pull/41)

i try your version of this library, still same behaviour

paulporto, 2020-11-01:
i just realize that i have to replace the rtc.c and rtc.h files in the cores. everything its working now thank you very much.

6v6gt-duino, 2020-11-19:
I've just hit exactly this problem. I had a development (a Bluepill based satellite clock) based on the Roger Clark STM32 core.
It used the RTC in the low power domain with a super cap on Vbat.
With the following library, the time survived power cycles without being reset : [https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/RTClock/src/RTClock.h](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/RTClock/src/RTClock.h)

After migrating recently to the official STM32 arduino core 1.9.0 and adapting the clock to use the STM32RTC library https://github.com/stm32duino/STM32RTC , the time was lost on a power cycle.

Anyway, I have applied the fix here and it now appears to work correctly again. That is the RTC time does not suffer an unwanted reset during a power cycle.
Thank you for fixing that.

This is what I did to apply the fix, in case it is useful for someone who wants to retrofit it into the current arduino core version 1.9.0 :

1. Removed just the src directory from the locally downloaded STM32RTC library e.g. ..../Arduino/libraries/STM32duino_RTC and replace it with the src directory obtained from [https://github.com/josh-gaby/STM32RTC](https://github.com/josh-gaby/STM32RTC) 
2. disabled any restrictions, requiring core release > 1.9.0 from the files in the source directory above. These are of the form: && (STM32_CORE_VERSION  > 0x01090000)

3. deleted rtc.h from e.g. ..../ArduinoData\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32
4. deleted rtc.c from e.g. ..../ArduinoData\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\stm32
5. deleted the include entry for rtc.h from  e.g. ....\\ArduinoData\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h

I don't doubt there is a more elegant way of achieving the same thing, but it worked for me. 
I'm looking forward to the next core release.

charliechancc, 2021-01-25:
> 
> 
> I've just hit exactly this problem. I had a development (a Bluepill STM32F103C8T6 based satellite clock) based on the Roger Clark STM32 core.
> It used the RTC in the backup domain, that is clock source LSE, with a super cap on Vbat.
> With the following library, the time survived power cycles without being reset : https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/RTClock/src/RTClock.h
> 
> After migrating some time later to the official STM32 arduino core 1.9.0 and adapting the clock to use the STM32RTC library https://github.com/stm32duino/STM32RTC , the time would not be retained during a power cycle.
> 
> Anyway, I have applied the fix here, developed by Josh-Gaby and it now appears to work correctly again. That is, the RTC time does not suffer an unwanted reset during a power cycle.
> 
> Many thanks for fixing that!
> 
> This is what I did to apply the fix, in case it is useful for someone who wants to retrofit it into the current arduino core version 1.9.0 :
> 
>     1. Removed just the src directory from the locally downloaded STM32RTC library e.g. ..../Arduino/libraries/STM32duino_RTC and replaced it with the src directory obtained from https://github.com/josh-gaby/STM32RTC
> 
>     2. disabled any restrictions, requiring core release > 1.9.0 from the files in the source directory above. These are of the form: && (STM32_CORE_VERSION  > 0x01090000)
> 
>     3. deleted rtc.h from e.g. ..../ArduinoData\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32
> 
>     4. deleted rtc.c from e.g. ..../ArduinoData\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\stm32
> 
>     5. deleted the include entry for rtc.h from  e.g. ....\ArduinoData\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h
> 
> 
> I don't doubt there is a more elegant way of achieving the same thing, but it worked for me.
> I'm looking forward to the next core release when these changes (or similar) will be rolled in.

Thanks! This method works on my custom STM32F103 circuit with supercap connected to VBAT pin.

Btw, for Windows user, the ArduinoData is located at "%USERPROFILE%\AppData\Local\Arduino15"

KiraVerSace, 2021-03-19:
> I've just hit exactly this problem. I had a development (a Bluepill STM32F103C8T6 based satellite clock) based on the Roger Clark STM32 core.
> It used the RTC in the backup domain, that is clock [...]

I do not know how to make the choice, there is no response from the official?

PR #58 (2021-12-16): Adds date retention in the backup memory for the stm32f1xx
This PR adds a RTC date retention for the stm32F1 MCUs. 
This new feature is saving/restoring the date (year, month, day, weekday) in the backup memory of he MCUs.
On the stm32F1, the Date is not present in the RTC peripheral but computed by the RTC driver.
The BackUp memory of the stm32F1 is powered by the VBat  when VDD is switched off.
It is not altered by the system Reset.

Each one-byte element of the RTC Date (DateToUpdate) is well placed in two contiguous 16bit backup registers (RTC_BKP_DATE, RTC_BKP_DATE + 1).  The registers ar [... cut]
closed 2021-12-16 by fpistm