stm32duino/STM32RTC #33: ClockSource : LSE_CLOCK is not working. Time not changing

arun912-ux, 2020-07-18:
Hi.
I'm having a problem with BLUE PILL RTC in LSE_CLOCK. Time is not changing. It keeps the same time.
Clock is ticking in LSI_CLOCK but resets when power is down.
 
My code is 

`
#include <STM32RTC.h>

/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

/* Change these values to set the current initial time */
const byte seconds = 0;
const byte minutes = 50;
const byte hours = 16;

/* Change these values to set the current initial date */
/* Monday 15th June 2015 */
const byte weekDay = 6;
const byte day = 18;
const byte month = 7;
const byte year = 20;

void setup()
{
  Serial.begin(115200);

  pinMode(PC13, OUTPUT);
  digitalWrite(PC13,0);
   
  // Select RTC clock source: LSI_CLOCK, LSE_CLOCK or HSE_CLOCK.
  // By default the LSI is selected as source.
  // rtc.setClockSource(STM32RTC::LSE_CLOCK);

  rtc.begin(); // initialize RTC 24H format
//
  // Set the time
  rtc.setHours(hours);
  rtc.setMinutes(minutes);
  rtc.setSeconds(seconds);

//  Set the date
  rtc.setWeekDay(weekDay);
  rtc.setDay(day);
  rtc.setMonth(month);
  rtc.setYear(year);

  // you can use also
  //rtc.setTime(8, 50, 0);         // hours, minutes, seconds
  //rtc.setDate(6, 18, 7, 20);       // weekday, day, month, year
}

void loop()
{
  // Print date...
  print2digits(rtc.getDay());
  Serial.print(" - ");
  print2digits(rtc.getMonth());
  Serial.print(" - ");
  print2digits(rtc.getYear());
  Serial.print("  ");

  // ...and time
  print2digits(rtc.getHours());
  Serial.print(":");
  print2digits(rtc.getMinutes());
  Serial.print(":");
  print2digits(rtc.getSeconds());
  Serial.print("   ");
  Serial.print(rtc.getWeekDay());
  Serial.println();
  
  digitalWrite(PC13,!digitalRead(PC13));
  
  delay(1000);
}

void print2digits(int number) {
  if (number < 10) {
    Serial.print("0"); // print a 0 before if the number is < than 10
  }
  Serial.print(number);
}
`

Help me please.

lucky62, 2020-08-05:
Hello, 
I observed the same issue - RTC is not working when source is LSE_CLOCK.

I am using the RTC_Tests (from STM32duino examples) with BluePill board.
LSI_CLOCK and HSE_CLOCK work, but LSE_CLOCK not - exactly not always.

I observed that when LSE is selected just after power up then it is not working.
When I first select the HSE then reset board by button and select LSE then it is working...

Serial Monitor output to show this behavior:
>16:39:50.242 -> Select clock Source:
16:39:50.242 -> 1- LSI_CLOCK
16:39:50.242 -> 2- LSE_CLOCK
16:39:50.242 -> 3- HSE_CLOCK
16:39:50.242 -> 
16:39:50.242 -> Enter number [1-3] 2
16:39:55.688 -> Test will use **LSE_CLOCK**
16:39:55.688 -> Testing asynchronous and synchronous prescaler setting
16:39:55.688 -> Async/Sync for default LSI clock: 0/0
**- - - - - - - - - - - - - - - - - - - - - - - - - - - - - LSE not working - reset after 40 seconds**
16:40:25.251 -> Select clock Source:
16:40:25.251 -> 1- LSI_CLOCK
16:40:25.251 -> 2- LSE_CLOCK
16:40:25.251 -> 3- HSE_CLOCK
16:40:25.251 -> 
16:40:25.251 -> Enter number [1-3] 2
16:40:32.918 -> Test will use **LSE_CLOCK**
16:40:32.918 -> Testing asynchronous and synchronous prescaler setting
16:40:32.918 -> Async/Sync for default LSI clock: 0/0
**- - - - - - - - - - - - - - - - - - - - - - - - - - - - - LSE still not working - reset**
16:41:17.253 -> Select clock Source:
16:41:17.253 -> 1- LSI_CLOCK
16:41:17.253 -> 2- LSE_CLOCK
16:41:17.253 -> 3- HSE_CLOCK
16:41:17.253 -> 
16:41:17.253 -> Enter number [1-3] 3
16:41:24.750 -> Test will use **HSE_CLOCK**
16:41:24.750 -> Testing asynchronous and synchronous prescaler setting
16:41:24.750 -> Async/Sync for default LSI clock: 0/0
16:41:24.750 -> Async/Sync for selected clock: 0/0
16:41:24.750 -> User Async/Sync set to 99/9999: 0/0
16:41:24.750 -> 08/05/20	15:06:22
16:41:25.746 -> 08/05/20	15:06:22
16:41:26.741 -> 08/05/20	15:06:23
16:41:27.737 -> 08/05/20	15:06:24
16:41:28.733 -> 08/05/20	15:06:25
16:41:29.728 -> 08/05/20	15:06:26
16:41:30.724 -> 08/05/20	15:06:27
16:41:31.720 -> 08/05/20	15:06:28
16:41:32.749 -> 08/05/20	15:06:29
16:41:33.744 -> 08/05/20	15:06:30
16:41:34.741 -> User Async/Sync reset use the computed one: 0/0
16:41:34.741 -> Testing date and time
16:41:34.741 -> 24H format, new year
16:41:34.741 -> 12/31/17	23:59:56
16:41:35.736 -> 12/31/17	23:59:56
16:41:36.730 -> 12/31/17	23:59:57
16:41:37.724 -> 12/31/17	23:59:58
16:41:38.720 -> 12/31/17	23:59:59
16:41:39.748 -> 01/01/18	00:00:00
16:41:40.744 -> 01/01/18	00:00:01
16:41:41.740 -> 01/01/18	00:00:02
16:41:42.735 -> 
16:41:42.735 -> Using Epoch API, set to Jan 1, 2016
16:41:42.735 -> Unix time = 1451606400
16:41:42.735 -> Seconds since Jan 1 2000 = 504921600
16:41:42.735 -> 01/01/16	00:00:00
16:41:43.730 -> Unix time = 1451606400
16:41:43.730 -> Seconds since Jan 1 2000 = 504921600
16:41:43.730 -> 01/01/16	00:00:01
[... 93 lines cut ...]
16:42:49.277 -> Unix time = 1451606406
16:42:49.277 -> Seconds since Jan 1 2000 = 504921606
16:42:49.277 -> 01/01/16	00:00:06
**- - - - - - - - - - - - - - - - - - - - - - - - - - - - - LSE is working - next reset**
16:42:54.332 -> Select clock Source:
16:42:54.332 -> 1- LSI_CLOCK
16:42:54.332 -> 2- LSE_CLOCK
16:42:54.332 -> 3- HSE_CLOCK
16:42:54.332 -> 
16:42:54.332 -> Enter number [1-3] 2
16:43:02.385 -> Test will use **LSE_CLOCK**
16:43:02.385 -> Testing asynchronous and synchronous prescaler setting
16:43:02.385 -> Async/Sync for default LSI clock: 0/0
16:43:06.403 -> Async/Sync for selected clock: 0/0
16:43:07.399 -> User Async/Sync reset use the computed one: 0/0
16:43:11.381 -> Testing date and time
16:43:11.381 -> 24H format, new year
16:43:12.742 -> 12/31/17	23:59:56
16:43:13.738 -> 12/31/17	23:59:56
16:43:14.733 -> 12/31/17	23:59:57
16:43:15.729 -> 12/31/17	23:59:58
16:43:16.725 -> 12/31/17	23:59:59
16:43:17.754 -> 01/01/18	00:00:00
16:43:18.749 -> 01/01/18	00:00:00
**- - - - - - - - - - - - - - - - - - - - - - - - - - - - - LSE is working...**

arun912-ux, 2020-08-05:
@lucky62 can you provide me your mail ID.

fpistm, 2020-08-05:
Hi all,

please @arun912-ux avoid to open several issue for the same thing. Thanks.
I will check that soon. Maybe there is a regression on F1, I need to test.

arun912-ux, 2020-08-05:
> Hi all,
> 
> please @arun912-ux avoid to open several issue for the same thing. Thanks.
> I will check that soon. Maybe there is a regression on F1, I need to test.

Sorry. It's my 1st time being on Github. I'll not do this again. 
Thanks.

FRASTM, 2020-08-27:
This situation is very difficult to reproduce on my environment.
There is an _Errata Sheet_ published by ST for [STM32F10xx8 STM32F10xxB](https://www.st.com/resource/en/errata_sheet/cd00190234-stm32f101x8b-stm32f102x8b-and-stm32f103x8b-mediumdensity-device-limitations-stmicroelectronics.pdf)
which applies to this device.
```
Workaround
It is recommended to mount an additional parallel feedback resistor (from 16 MΩ to 22 MΩ) on board to help the oscillation start-up in all cases (see Figure 1).``` 

![ES096_LSE start-up in harsh environments](https://user-images.githubusercontent.com/51417392/91419790-1d9dd000-e854-11ea-9d3b-e4488f8973ee.png)

Could the workaround help fixing this issued ?

FRASTM, 2020-09-01:
another sw approach could be to switch LSE clock OFF and ON until the flag LSE RDY is set, as follows:
```
int timeout = LSE_TIMEOUT_VALUE;
      if (__HAL_RCC_GET_FLAG(RCC_FLAG_LSERDY) == RESET) {

        __HAL_RCC_LSE_CONFIG(RCC_LSE_ON);
        /* loop switching LSE on again */
        while (__HAL_RCC_GET_FLAG(RCC_FLAG_LSERDY) !=1) {
          timeout--;
          if (timeout == 0) {
                  __HAL_RCC_LSE_CONFIG(RCC_LSE_OFF);
                  timeout = LSE_TIMEOUT_VALUE;
                  __HAL_RCC_LSE_CONFIG(RCC_LSE_ON);
          }
        }
```

fpistm, 2021-04-22:
This issue was not reproduced.
No feedback from OP about @FRASTM comments.
So I close it.

closed 2021-04-22 by fpistm