stm32duino/STM32RTC #45: HOUR_FORMAT_12 is undefine?

KiraVerSace, 2020-12-17:
OK, I use it in the vscode with platformio, I copy all the files in to the root.
After compile it, it report's error.
There is two ways to slove it, but I don't know which is the best solution?
1. remove the rtc.c and rtc.h in your library, it will search it in the .platformio/packages/framework-arduinoststm32/libraries/SrcWrapper/src/stm32/rtc.c
2. modify the file in your library change all the [FIXME: STM32_CORE_VERSION  > 0x01090000 to STM32_CORE_VERSION  >= 0x01090000]
so what I should do, I compare the two files rtc.c , the difference is with the funtion static inline int _log2(int x),
Could you help me?

fpistm, 2020-12-17:
Hi @KiraVerSace 
No one is the good solution. The rtc driver has been moved from the core to the library. 
My guess is that you used the PIO library manager which does not take in account if a library version is release or not.
The 1.1.0 referenced by PIO is not officially released so use the latest official release of this library.

closed 2020-12-17 by fpistm

KiraVerSace, 2020-12-17:
> Hi @KiraVerSace
> No one is the good solution. The rtc driver has been moved from the core to the library.
> My guess is that you used the PIO library manager which does not take in account if a library [...]

As you mentioned I use the library from here which is you released it.
So I have to change the STM32_CORE_VERSION or it will report the error cause there has two rtc.c and rtc.h file in the directory?

Or could you tell me the right way to use your library under the platformIO.

HaHa, Thanyou very much!

fpistm, 2020-12-17:
> Or could you tell me the right way to use your library under the platformIO.

I never use PIO so can't help on this.
See #30, #31 ...