stm32duino/STM32RTC #49: The 1.1.0 version in the PlatformIO library manager isn't the same as the repository 1.1.0

wilmardo, 2021-06-07:
[image]

As seen in the screenshot (bottom right) the PlatformIO library manager seems to think that the 1.1.0 tag has been released May 2020.

This currently gives issues with the latest ststm32 platform (14.0.0) since the RTC files have been moved from the core to this plugin but the plugin is stuck on the bogus 1.1.0 release of May 2020 and is missing these moved RTC files.

The date of the release seems to be the same as the commit that bumped the version in `library.properties`
1e99a58398ee9c7c9f564e609d811a06128826c4

fpistm, 2021-06-07:
Hi,
I don't know how PIO library version are managed. 
Anyway, I've already had several issue before because PIO considered as 1.1.0 the master of this repo while the 1.1.0 was not released.
You should open an issue on PIO side.

closed 2021-06-07 by fpistm

fpistm, 2021-06-07:
About this

> The date of the release seems to be the same as the commit that bumped the version in `library.properties`
> [1e99a58](https://github.com/stm32duino/STM32RTC/commit/1e99a58398ee9c7c9f564e609d811a06128826c4)

Even if the library.version has been bumped in the library.properties this does not meansit has been released while the tag is used for the official release.