stm32duino/STM32RTC #50: PlatformIO library registry has wrong version

maxgerhardt, 2021-06-16:
See [community topic](https://community.platformio.org/t/stm32-new-update-error-on-rtc-library/22017/6?u=maxgerhardt).

When attempting to use this library with the latest [14.0.0 platform-ststm32](https://github.com/platformio/platform-ststm32/releases) release, which utilizes STM32Duino core v2.0.0, via doing e.g.

```ini
[env:genericSTM32F103RE]
platform = ststm32
board = genericSTM32F103RE
framework = arduino
lib_deps =
     stm32duino/STM32duino RTC @ ^1.1.0
```

one runs into tons of compilation failures (with the [SimpleRTC.ino](https://github.com/stm32duino/STM32RTC/blob/master/examples/SimpleRTC/SimpleRTC.ino))

```
.pio\libdeps\genericSTM32F103RE\STM32duino RTC\src/STM32RTC.h:77:17: error: 'HOUR_FORMAT_12' was not declared in this scope
   77 |       HOUR_12 = HOUR_FORMAT_12,
      |                 ^~~~~~~~~~~~~~
.pio\libdeps\genericSTM32F103RE\STM32duino RTC\src/STM32RTC.h:78:17: error: 'HOUR_FORMAT_24' was not declared in this scope
   78 |       HOUR_24 = HOUR_FORMAT_24
      |                 ^~~~~~~~~~~~~~
```
etc. 

this is because the version that is stored in the PlatformIO registry (https://platformio.org/lib/show/5502/STM32duino%20RTC/installation) is indeed tagged as 1.1.0 but when downloading it, it is apparent that the contents are different to what is released here as v1.1.0. Also the release dates are different.

PlatformIO:

[image]

This repo: Released 2 months ago, [22th April 2021](https://github.com/stm32duino/STM32RTC/releases/tag/1.1.0) [not 1 year ago].

When doing `pio lib -g install "stm32duino/STM32duino RTC @ ^1.1.0"` one can see that certain files are missing

PIO version

```
\---src
        STM32RTC.cpp
        STM32RTC.h
```

Repo's 1.1.0 version:  Has [rtc.h and rtc.h additionally](https://github.com/stm32duino/STM32RTC/tree/1.1.0/src). 

Please re-upload / re-publish the 1.1.0 library version in the PlatformIO library registry with the *actual* 1.1.0 library code.

fpistm, 2021-06-17:
Hi @maxgerhardt 

> Please re-upload / re-publish the 1.1.0 library version in the PlatformIO library registry with the _actual_ 1.1.0 library code.

we are not responsible of the PIO library registry. We already get such issue open here but PIO seems used the library.properties version without check if a tag exists. In Arduino ecosystem a new library version is proposed only whan there is a new tag an release. Ask to PIO to fix this.

closed 2021-06-17 by fpistm

maxgerhardt, 2021-06-17:
@ivankravets could you please delete the 1.1.0 version of https://platformio.org/lib/show/5502/STM32duino%20RTC/ and replace it with https://github.com/stm32duino/STM32RTC/releases/tag/1.1.0?

fpistm, 2021-06-17:
Note that it is the same issue for other lib (STM32LowPower at least)

ivankravets, 2021-06-22:
Hi there,

The libs mentioned above were re-updated with their tagged versions.

Thanks!

fpistm, 2021-06-22:
Thank you @ivankravets