stm32duino/STM32RTC #52: Error build in STM32Duino under platformio (VSCODE)

rozrabiak, 2021-07-03:
Errors when building under STM32 ST 14.0.1 and 14.0.0. 
Under 13.0.0 building with no errors.

.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:90:28: error: 'SS_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                            ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:90:37: error: 'MM_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                                     ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:90:46: error: 'HH_MSK' was not declared in this scope
   90 |       MATCH_HHMMSS       = SS_MSK | MM_MSK | HH_MSK,         // Every Day
      |                                              ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:91:28: error: 'SS_MSK' was not declared in this scope
   91 |       MATCH_DHHMMSS      = SS_MSK | MM_MSK | HH_MSK | D_MSK, // Every Month
      |                            ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:91:37: error: 'MM_MSK' was not declared in this scope
   91 |       MATCH_DHHMMSS      = SS_MSK | MM_MSK | HH_MSK | D_MSK, // Every Month
      |                                     ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:91:46: error: 'HH_MSK' was not declared in this scope
   91 |       MATCH_DHHMMSS      = SS_MSK | MM_MSK | HH_MSK | D_MSK, // Every Month
      |                                              ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:91:55: error: 'D_MSK' was not declared in this scope
   91 |       MATCH_DHHMMSS      = SS_MSK | MM_MSK | HH_MSK | D_MSK, // Every Month
      |                                                       ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:94:28: error: 'SS_MSK' was not declared in this scope
   94 |       MATCH_MMDDHHMMSS   = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK,
      |                            ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:94:37: error: 'MM_MSK' was not declared in this scope
   94 |       MATCH_MMDDHHMMSS   = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK,
      |                                     ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:94:46: error: 'HH_MSK' was not declared in this scope
   94 |       MATCH_MMDDHHMMSS   = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK,
      |                                              ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:94:55: error: 'D_MSK' was not declared in this scope
   94 |       MATCH_MMDDHHMMSS   = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK,
      |                                                       ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:94:63: error: 'M_MSK' was not declared in this scope
   94 |       MATCH_MMDDHHMMSS   = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK,
      |                                                               ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:28: error: 'SS_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                            ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:37: error: 'MM_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                                     ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:46: error: 'HH_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                                              ^~~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:55: error: 'D_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                                                       ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:63: error: 'M_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                                                               ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:95:71: error: 'Y_MSK' was not declared in this scope
   95 |       MATCH_YYMMDDHHMMSS = SS_MSK | MM_MSK | HH_MSK | D_MSK | M_MSK | Y_MSK
      |                                                                       ^~~~~
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h: In member function 'bool STM32RTC::isTimeSet()':
.pio\libdeps\blackpill_f401cc\STM32duino RTC\src/STM32RTC.h:207:14: error: 'RTC_IsTimeSet' was not declared in this scope; did you mean 'isTimeSet'?  
  207 |       return RTC_IsTimeSet();
      |              ^~~~~~~~~~~~~
      |              isTimeSet
*** [.pio\build\blackpill_f401cc\src\main.cpp.o] Error 1

rozrabiak, 2021-07-03:
http://www.wklejto.pl/907208

fpistm, 2021-07-04:
You probably don't have the proper RTC library. 
RTC drivers has been moved from the core to the library. 
Moreover I know PIO package wrong version of this library and have reindex it properly 2 weeks ago.
See https://github.com/stm32duino/STM32RTC/issues/50.

There is no issue with Arduino IDE as it deals properly with the version.
So I close it as only linked to PIO.

closed 2021-07-04 by fpistm