stm32duino/STM32RTC #53: Unix time alarm

le0pa-RD, 2021-07-28:
Hello, is it possible to set an alarm to an unix time with this library?

ABOSTM, 2021-08-02:
Hi @le0pa-RD,
You can have a look at the readme https://github.com/stm32duino/STM32RTC/blob/master/README.md
There are some API like:
```
void setEpoch(uint32_t ts, uint32_t subSeconds = 0)
void setAlarmEpoch(uint32_t ts, Alarm_Match match = MATCH_DHHMMSS, uint32_t subSeconds = 0)
...
```
I hope this match with your unix time definition

le0pa-RD, 2021-08-03:
Hello @ABOSTM,
what I want to do is to set an alarm to a UNIX timestamp (SECONDS SINCE JAN 01 1970) instead of the default timestamp (DHHMMSS) but I think there is no such function in this library.
If I'm wrong please let me know.

ABOSTM, 2021-08-04:
I think  you can simply make a subtraction knowing that time reference in this libary (epoch) is hardcoded to 1st January 2000, 00:00:00
https://github.com/stm32duino/STM32RTC/blob/4873e0b2ee2074bf3fff080e308403e03afcac7d/src/STM32RTC.cpp#L41

So you could convert unix time to RTC library time, by subtracting the number of seconds from 1970 to 2000 

But be careful, the  STM32RTC alarm (the hardware peripheral) doesn't support year and month (it is supported for time, but not for alarm)
https://github.com/stm32duino/STM32RTC/blob/4873e0b2ee2074bf3fff080e308403e03afcac7d/src/STM32RTC.cpp#L765

It means your alarm will be set for every month and every year.
Let say you program alarm for December 1st of 2022 it will expire the Septembre 1st of 2021, the  October 1st of 2021, ...

closed 2021-08-09 by fpistm