stm32duino/STM32RTC #64: rtc.setClockSource(STM32RTC::LSE_CLOCK); not working

madhavanatmel, 2022-04-14:
HI, When i enabling the rtc.setClockSource(STM32RTC::LSE_CLOCK); both Black pill and blue pill are getting hanging . 
The simpleRTC example code working with following selection .

default  : rtc.setClockSource(STM32RTC::LSI_CLOCK); 

 rtc.setClockSource(STM32RTC::HSE_CLOCK); 

not working on 
rtc.setClockSource(STM32RTC::LSE_CLOCK);

ABOSTM, 2022-04-14:
Hi @madhavanatmel ,
I tested on my bluepill and it is working fine with LSE.
Core version: 2.2.0
RTC library version: 1.2.0
Variant: BluePill F103C8
Sketch: see below

Can you test the same core version and library version?
Can you confirm the variant you are using ?
Can check that you don't have a fake bluepill (and specially that you have LSE on board) 
see https://embeddedtronicsblog.wordpress.com/2018/12/29/fake-stm32-blue-pill-boards/

```
/*
  SimpleRTC

  This sketch shows how to configure the RTC and to display
  the date and time periodically

  Creation 12 Dec 2017
  by Wi6Labs
  Modified 03 Jul 2020
  by Frederic Pillon for STMicroelectronics

  This example code is in the public domain.

  https://github.com/stm32duino/STM32RTC
*/

#include <STM32RTC.h>

/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

/* Change these values to set the current initial time */
const byte seconds = 0;
const byte minutes = 0;
const byte hours = 16;

/* Change these values to set the current initial date */
/* Monday 15th June 2015 */
const byte weekDay = 1;
const byte day = 15;
const byte month = 6;
const byte year = 15;

void setup()
{
  Serial.begin(9600);
 Serial.println("ABO LSE");
  // Select RTC clock source: LSI_CLOCK, LSE_CLOCK or HSE_CLOCK.
  // By default the LSI is selected as source.
  //rtc.setClockSource(STM32RTC::LSE_CLOCK);
  rtc.setClockSource(STM32RTC::LSE_CLOCK);

  rtc.begin(); // initialize RTC 24H format

  // Set the time
  rtc.setHours(hours);
  rtc.setMinutes(minutes);
  rtc.setSeconds(seconds);

  // Set the date
  rtc.setWeekDay(weekDay);
  rtc.setDay(day);
  rtc.setMonth(month);
  rtc.setYear(year);

  // you can use also
  //rtc.setTime(hours, minutes, seconds);
  //rtc.setDate(weekDay, day, month, year);
}

void loop()
{
  // Print date...
  Serial.printf("%02d/%02d/%02d ", rtc.getDay(), rtc.getMonth(), rtc.getYear());

  // ...and time
  Serial.printf("%02d:%02d:%02d.%03d\n", rtc.getHours(), rtc.getMinutes(), rtc.getSeconds(), rtc.getSubSeconds());

  delay(1000);
}
```

madhavanatmel, 2022-04-14:
HI, Both boards are Genuine board and using latest Version as you mentioned . but i think  the Good Point is LSE Crystal Working condition . if it's working with you then following phenomenon must be the reason . i have used PC14 & PC15 (without rtc function )pin for input purpose before some time in blue & black board  . LSE crystal got failure that time . i will change the LSE crystal and i will confirm You .

madhavanatmel, 2022-04-15:
Hi, after LSE crystal rtc timer started working without battery condition .

with battery connection  rtc not keeping time and  the movement battery is connected to VB then output print time is same value i mean timer giving the same time value again and again .

madhavanatmel, 2022-04-17:
hi, Interesting ! ,fixed branded Battery - Panasonic cell  . Finally RTC started working with battery and keeping rtc time .however i kept 24 + 6 =30 hours for observation . Seems RTC much more sensitive than main codes .

closed 2022-04-17 by madhavanatmel

augustinenwafor1997, 2022-04-26:
> Hi, after LSE crystal replacement rtc timer started working without battery condition .
> 
> with battery connection rtc not keeping time and the movement battery is connected to VB then output print time is same value i mean timer giving the same time value again and again .

what crystal did you change to?

fpistm, 2022-04-26:
> I am facing this issue with my blackpill. My code works fine on the blue pill but freezes on the blackpill. I am curious if there is any lasting solution to this problem

As stated it is an hardware issue. Crystal was faulty. I've a BlackPill an it works as expected but I saw here or on other forums several people claiming about BAS LSE crystal mounted on BlackPill.