stm32duino/STM32RTC #66: STM32F103C8T6 RTC losing about one second on each system reset.

6v6gt-duino, 2022-05-22:
I'm using the RTC of a Bluepill STM32F103C8T6. It has a supercap on VBat. The installed version of the STM32duino rtc library is 1.2.0.
The STM32 Arduino Core is at version 2.2.0.

If I press the reset button on the Bluepill, the RTC appears to lose about 1 second on each press.

I am using the following code based on this example to demonstrate this problem:  https://github.com/stm32duino/STM32RTC/blob/main/examples/SimpleRTC/SimpleRTC.ino

```
/*
  SimpleRTC
  This sketch shows how to configure the RTC and to display
  the date and time periodically
  Creation 12 Dec 2017
  by Wi6Labs
  Modified 03 Jul 2020
  by Frederic Pillon for STMicroelectronics
  This example code is in the public domain.
  https://github.com/stm32duino/STM32RTC
*/

#include <STM32RTC.h>

/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

/* Change these values to set the current initial time */
const byte seconds = 0;
const byte minutes = 53;
const byte hours = 12;

/* Change these values to set the current initial date */
const byte weekDay = 1;
const byte day = 22;
const byte month = 5;
const byte year = 22;

void setup()
{
  Serial.begin(115200);

  pinMode(PB12, INPUT_PULLUP ) ;

  // Select RTC clock source: LSI_CLOCK, LSE_CLOCK or HSE_CLOCK.
  // By default the LSI is selected as source.

  rtc.setClockSource(STM32RTC::LSE_CLOCK);

  rtc.begin(); // initialize RTC 24H format

}

void loop()
{

  if ( ! digitalRead(PB12) ) {
    // Set the time
    rtc.setHours(hours);
    rtc.setMinutes(minutes);
    rtc.setSeconds(seconds);

    // Set the date
    rtc.setWeekDay(weekDay);
    rtc.setDay(day);
    rtc.setMonth(month);
    rtc.setYear(year);
    Serial.println( "reset time" ) ;
    delay( 100 ) ;
  }

  // Print date...
  Serial.printf("%02d/%02d/%02d ", rtc.getDay(), rtc.getMonth(), rtc.getYear());

  // ...and time
  Serial.printf("%02d:%02d:%02d.%03d\n", rtc.getHours(), rtc.getMinutes(), rtc.getSeconds(), rtc.getSubSeconds());

  delay(1000);
}
```
I have set the clock and taken a series of screen shots and  have logged the results. The result is that the delta between local (PC) time and the Bluepill RTC time increases by about one second for each reset. These seconds are just lost.

```
local time	rtc time     delta (seconds)
12.57.05	12:53:47     198s    
13:00:39	12:57:21     198s  
13:03:08    	12:59:49     199s
13:13:09	13:09:51     198s	
13:18:35        13:15:16     199s

3 X reset here

13:22:16	13:18:55     201s  
13:26:16	13:22:55     201s
13:29:21	13:26:00     201s

5 x reset here

13:35:49	13:32:24     205s
13:38:58	13:35:33     205s
```

As can be seen, the delta increases after each series of resets.

This appears similar to this case:
https://community.st.com/s/question/0D50X00009XkgBWSAZ/stm32-rtc-loses-one-second-after-each-reset

6v6gt-duino, 2022-06-02:
Thank you very much for responding.
I grabbed the 4 files from your pull request #67 and tested these together with my application using a Bluepill (the same setup as in the original Issue).
It appears to work for the tests I applied, that is synchronizing the time automatically with a known time source (DCF77), repeatedly resetting the device, then checking again a similarly accurate time source before the application again found the DCF77. I saw no noticeable loss of time on switch over to the RTC. I repeated the exercise 3 times. Further, I saw no compiler warnings from the new code.
So, in summary, it does indeed appear to work. Many thanks for dealing with that.

ABOSTM, 2022-06-03:
@6v6gt-duino , thanks for the feedback.
Issue will be closed automatically once PR will be merged.

PR #67 (2022-06-06): refactor: Improve reliability and simplify implementation
refactor: Improve reliability and simplify implementation

* call HAL_RTC_Init() only when needed to avoid losing hundreds
  of milliseconds on RTC counter
* remove useless check of _configured at the beginning of
  STM32RTC class methods
* Use flag "INITS" to detect whether RTC calendar is initialized
  (except for F1 which doesn't have this flag)
  This prevent the use of year 2000 which is the default hardware
  reset value and doesn't trigger INITS flag
* Use Backup registers only for F1, and no more for other series
* Simplify clas [... cut]
closed 2022-06-06 by fpistm