stm32duino/STM32RTC #74: Add simultaneous support for RTC_ALARM_A and RTC_ALARM_B

batkov, 2022-08-14:
Hi,

I'm trying to run few RTC Alarms on one chip simultaneously. 
I've found that `RTC_ALARM_A` (rtc.c) is used by default and I see no way to run it along with `RTC_ALARM_B` on chip supporting such a feature.

It seems like `RTC_HandleTypeDef RtcHandle`  initialized once and would be able to run only one type of alarm at one point of time.

Can you add support for `RTC_ALARM_B`?
Can you suggest how to add such a support?

fpistm, 2022-08-22:
Hi @batkov 
sorry for the delay (was in vacations).
You're right, RTC_ALARM_B is currently not supported as this library was designed originally to be compliant with Arduino RTCZero library which only support one alarm IIRW. Moreover not all STM32 series have a second alarm.
Anyway, it should be fine to add it. Unfortunately I cannot give a date for this. Any help are welcome.
Maybe a way to add it would be to check if we can extend the alarm API's to select the alarm with RTC_ALARM_A by default to keep backward compatibility and of course available only for STM32 series with RTC_ALARM_B support.

batkov, 2022-08-22:
Yes, I already implemented something similar to what you've mentioned. Will fix backward compatibility issues and create PR soon. 
I have a question though. I've changed RTC methods to receive alarm type, but kept old ones for compatibility.
I've added `v2` prefixes for new methods. Probably, @fpistm, you'll have netter ideas how to name those methods.

fpistm, 2022-08-23:
Well for example you can do something like that (not tested):
```
    uint32_t getAlarmSubSeconds(uint32_t alarm = RTC_ALARM_A);
    uint8_t getAlarmSeconds(uint32_t alarm = RTC_ALARM_A );
    uint8_t getAlarmMinutes(uint32_t alarm = RTC_ALARM_A );
    uint8_t getAlarmHours(AM_PM *period = nullptr, uint32_t alarm = RTC_ALARM_A);
    uint8_t getAlarmHours(uint32_t alarm = RTC_ALARM_A);
```

batkov, 2022-08-24:
@fpistm please check https://github.com/stm32duino/STM32RTC/pull/75

fpistm, 2022-08-26:
Hi @batkov 
I will review it soon.
As a first feedback I would not use v2_ suffix.

batkov, 2022-08-26:
Hi @fpistm ,
Yeah, but I did not come with anything better. Please propose something, and I'll update the PR.

JeroenIoT, 2022-10-09:
Hi @batkov,

Would it be an idea to add 1 extra function to set the alarm which we are configuring, setting and getting (by default this is alarmA). This function would only be valid for number of processors capable of handling alarm A / B.

- Imho changes to the following functions:

```
void RTC_StartAlarm(uint8_t day, uint8_t hours, uint8_t minutes, uint8_t seconds, uint32_t subSeconds, hourAM_PM_t period, uint8_t mask) {
   |
if (alarmtype == ALARM_B)
{ 	RTC_AlarmStructure.Alarm = RTC_ALARM_B;}
else
{ 	RTC_AlarmStructure.Alarm = RTC_ALARM_A; }
   |
}
```
At a first glance the following to be changed to handle both alarmA and alarmB (depending on alarmType set by used:

- void RTC_StopAlarm(void)
- bool RTC_IsAlarmSet(void)

PR #86 (2023-01-18): feat: add second alarm (B) support
Supersede #75

Fixes #74 

### Test
- [x] STM32F0
- [x] STM32F1
- [x] STM32F2
- [x] STM32F3
- [x] STM32F4
- [x] STM32F7
- [x] STM32G0
- [x] STM32G4 
- [x] STM32H7
- [x] STM32L0
- [x] STM32L1
- [x] STM32L4
- [x] STM32L5
- [x] STM32U5
- [x] STM32WB
- [x] STM32WL

Sketch used to test added to the examples folder: `RTCReset.ino`
commit d5484bbf (2023-01-18): feat: add second alarm (B) support

Fixes #74
closed 2023-01-18 by fpistm