stm32duino/STM32RTC #78: One Second Interrupt not working on Nucleo G071RB

harji2130, 2022-11-01:
I am using an STM32 RTC_Seconds example to test one second interrupt on my Nucleo G071RB (STM32G071RB) board but the interrupt does not work on my Nucleo board. I also check on Agafia SG0 (STM32G071CBT6) but the second interrupt does not work.

On looking into the source files, the STM32G71x MCU has OneSecond_IRQn defined in **rtc.h** as
`#define ONESECOND_IRQn RTC_TAMP_IRQn`

 that uses global RTC/TAMP interrupt
` RTC_TAMP_IRQn               = 2,      /*!< RTC interrupt through the EXTI line 19 & 21`   defined in stm32g071xx.h file

ABOSTM, 2022-11-07:
Hi @harji2130,
Thanks for reporting this issue, you are right there is a bug especially for STM32G0 implementatio,
which is slightly different from other STM32 series.

I proposed a fix : #79
Fell free to test it.

drucknase, 2022-11-10:
Hi,
I have a similar effect with the STM32L073LZ. So the RTC second example is not working. Can you confirm? When will the new version as proposed on fix #79 be available?

fpistm, 2022-11-10:
> Hi, I have a similar effect with the STM32L073LZ. So the RTC second example is not working. Can you confirm? When will the new version as proposed on fix #79 be available?

The fix only apply for G0. @ABOSTM could you check if the same is valid for L0, please?

fpistm, 2022-11-10:
All series tested. Thanks @ABOSTM .

PR #79 (2022-11-14): fix: 1 RTC vector for both HAL IRQHandler Alarm and WakeUpTimer
fix: 1 RTC vector for both HAL IRQHandler Alarm and WakeUpTimer

Sometimes, both HAL IRQ handler should be call from RTC vector
* HAL_RTC_AlarmIRQHandler()
* HAL_RTCEx_WakeUpTimerIRQHandler()

Fixes #78
closed 2022-11-14 by fpistm