stm32duino/STM32RTC #85: LSI not counting time while powered off, LSE backupdomain not working at all

Zelsof, 2022-12-29:
_Hello, 
I know that parts of this issue have been already discussed and I am sorry for repeating some. Unfortunately I couldn't find a working solution for my problem. Hopefully anybody can help. Hardwares-setup: I am using an STM32 F446RE board. Vbat is supplied by 3V battery._ 

**LSI CLOCK**
(no clocktype setup, or use of command `RTC.setClockSource(STM32RTC::LSI_CLOCK);`):
- the RTCworks, as long the board is powered 
- accuracy is tolerable but not perfect (~2 seconds of deviation after a few minutes)
- RTC clock source selection in RCC_BDCR is set `10: LSI oscillator clock used as the RTC clock` (seems good)
- when board is not powered for 5 minutes and then powered again:
     * the RTC continues with the last time, before being switched off (the 5 minutes never existed)
     *  `RTC.isTimeSet()` returns `true`at startup
     * so backup domain works generally and is kept on power by battery, BUT the RTC is no continuing to count the time.
   
**LSE CLOCK**
(by `RTC.setClockSource(STM32RTC::LSE_CLOCK);`)
- the RTC works, as long as the board is powered
- accuracy seems to be quiet good
- RTC clock source selection in RCC_BDCR is set `01: LSE oscillator clock used as the RTC clock` (seems good)
- once the board is powered off and on again, 
    * all RTC information are lost
    * `RTC.isTimeSet()` returns `false` at startup
    * backup domain seems not to be working at all (at least at startup I can prove that)

**HSE CLOCK**
(by `RTC.setClockSource(STM32RTC::HSE_CLOCK);`)
- not working, since no HSE oscillator -> CPU stucks when power on

_I want to use the LSE clock, but it doesn't keep anything in the backupdomain, whereas LSI does, but not counting time.
Does anybody has an idea what is wrong, and how I can realize a working setup?_

_Thank you in advance!_

------------------------------------------------------------------------
**Additionally here are a few, maybe relevant register entries:**
```
RCC_CR(clock control register):

HSION - 1: HSI oscillator ON
HSIRDY - 1: HSI oscillator ready
HSITRIM - 0x10
HSICAL - 0x73
HSEON - 0: HSE oscillator OFF
HSERDY - 0: HSE oscillator not ready
HSEBYP - 0: HSE oscillator not bypassed
CSSON - 0: Clock security system OFF (Clock detector OFF)
```

```
RCC_PLLCFGR(PLL configuration register):

PLLM - 001000: PLLM = 8
PLLN - 010110100: PLLN = 180 
PLLP - 00: PLLP = 2
PLLSRC - 0: HSI clock selected as PLL and PLLI2S clock entry
PLLQ - 0010: PLLQ = 2
```

```
RCC_CFGR(clock configuration register):

SW - 10: PLL_P selected as system clock
SWS - 10: PLL used as the system clock
HPRE - 0: system clock not divided
PPRE1 - 101: AHB clock divided by 4
PPRE2 - 100: AHB clock divided by 2
RTCPRE - 00000: no clock
MCO - 00: HSI clock selected
MCO1PRE - 0: no division
MCO2PRE - 0: no division
MCO2 - 00: System clock (SYSCLK) selected
```

```
RCC_BDCR(Backup domain control register):

LSEON - 1: LSE clock ON
LSERDY - 1: LSE clock ready
LSEBYP - 0: LSE oscillator not bypassed
LSEMOD - 0: LSE oscillator "low power" mode selection
RTCSEL - 10: LSI oscillator clock used as the RTC clock
BDRST - 0: Reset not activated
```

```
RCC_CSR(clock control & status register):

LSION - 1: LSI RC oscillator ON
LSIRDY - 1: LSI RC oscillator ready
[...]
```

```
RTC_PRER(RTC prescaler register):

PREDIV_S - PREDIV_S = 249
PREDIV_A - PREDIV_A = 127
```

fpistm, 2022-12-29:
Hi. Which board exactly?
How vbat is connected?
Environment: ide version, core, libraries, sketch,...?

Zelsof, 2022-12-29:
Hi, thanks for your reply.
Board version is F446RET (MB1136 C-04)
I have a case for 2 AA batteries. + connected to Vbat, - connected to ground.

I use platform.io with stm32 platform on arduino framework. TM32duino RTC version is latest (1.3.4).

fpistm, 2022-12-30:
Pio is not up to date with core version. And are you sure you used latest RTC library version.

Zelsof, 2022-12-30:
I just updated everything. PlatformIO, Arduino_Core_STM32 (framework-arduinoststm32 - 4.20200.221104), RTC (1.3.4) and the ST STM32 Platform are now all up-to-date.

But the behaviour is still the same:
LSI: not counting on the time, when board is powered off
LSE: backup domain doesn't save anything

fpistm, 2022-12-30:
As far as I know pio is not up to date. An issue is open for that. So even if you update pio st stm32 you do not use latest core version. I will try next week with arduino IDE.

Zelsof, 2022-12-30:
Thanks. I just tried the workaround with version 2.4.0, but it didn't change anything. I am still hoping for help :)